muccc/gr-iridium #194: rtlsdr v3 not working

m1cdq, 2024-05-24:
I have a v3 and version v4 of the rtlsdr.

the v4 works fine. but i get nothing on the v3

when i try the v3  on its own. it is detected by SoapySDRUtil --probe 

run the command 
iridium-extractor -D 4 /usr/src/gr-iridium/examples/rtl-sdr.conf > output.bits

nothing detected  

swap out for the v4  again it detected   and when i run 
iridium-extractor -D 4 /usr/src/gr-iridium/examples/rtl-sdr.conf > output.bits
i get data 

the V3 is working fine   using sdr app in linux and windows. also checked the bias-t voltage is there  with a meter

alphafox02, 2024-05-26:
Caring over from email conversation, I've finally got around to hooking up my v3 and do see what seems to be the same issue. More context for the conversation, I'm using 22.04 w/ librtlsdr

https://github.com/osmocom/rtl-sdr/releases/tag/v0.6.0

I've tried both gr-osmosdr source and soapy source within gr-iridium. I've tried internal and confirmed working external bias-t and get zero results in my quick test

```
1716722619 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1716722620 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
1716722621 | i:   0/s | i_avg:   0/s | q_max:    0 | i_ok:   0% | o:    0/s | ok:   0% | ok:   0/s | ok_avg:   0% | ok:          0 | ok_avg:   0/s | d: 0
```
By simply changing out the v3 to the RTLSDR v4 I get at least something

```
1716722855 | i:   7/s | i_avg:   1/s | q_max:    3 | i_ok:  87% | o:   11/s | ok:  87% | ok:   6/s | ok_avg:  87% | ok:          7 | ok_avg:   1/s | d: 0
1716722856 | i:  15/s | i_avg:   4/s | q_max:    2 | i_ok:  93% | o:   23/s | ok:  93% | ok:  14/s | ok_avg:  91% | ok:         22 | ok_avg:   3/s | d: 0
```
Checking with both SDR++/CublicSDR on the same freq that gr-iridium looks at, I can see what appear to be bursts with the v3 using either the internal bias-t (setting on) and an external bias-t.

I guess next perhaps I need to try and earlier or later release of librtlsdr?

alphafox02, 2024-05-26:
Just confirmed that be moving to the latest release of librtlsdr brings back results with the v3. Be aware I took the extra step of changing the .so filename as it jumped from 0.6.0 to a 2.x name and I didn't feel like breaking anything. I would back up what I'm overwriting just in case though. I'll continue to test this and make a new .deb package for install. 

https://github.com/osmocom/rtl-sdr/releases/tag/v2.0.1

unzip rtl-sdr-2.0.1.zip 
cd rtl-sdr-2.0.1/
mkdir build && cd build
cmake ..
make
sudo mv src/librtlsdr.so.2.0.1 /usr/lib/x86_64-linux-gnu/librtlsdr.so.0.6.0 
sudo mv src/rtlsdr_* /usr/bin/

alphafox02, 2024-05-26:
Thanks for pointing this out.

gprime31, 2025-07-08:
> Just confirmed that be moving to the latest release of librtlsdr brings back results with the v3. Be aware I took the extra step of changing the .so filename as it jumped from 0.6.0 to a 2.x name [...]

Life Saver, thank you!!!

For the Pi running the rtl-sdr blog v4:

cd rtl-sdr-blog
sudo rm -rvf /usr/lib/librtlsdr* /usr/include/rtl-sdr* /usr/local/lib/librtlsdr* /usr/local/include/rtl-sdr* /usr/local/include/rtl_* /usr/local/bin/rtl_*
sudo cp src/librtlsdr.so.0.6git /usr/lib/aarch64-linux-gnu/librtlsdr.so.0.6.0
sudo cp src/rtlsdr_* /usr/bin/