muccc/gr-iridium #198: Error: Iridium Sniffer Setup Issue with gr-iridium and HackRF on Ubuntu 22.04

ashwini1890, 2024-07-25:
I'm new to gr-iridium and the Iridium Toolkit, attempting to build my own Iridium sniffer using Ubuntu 22.04 and HackRF. Following recommendations, I've installed GNU Radio 3.10. When I ran the command sudo iridium-extractor -D 4 examples/hackrf.conf > output.bits to sniff packets, I encountered an error. Could someone please explain the reason for this error and help me resolve it?

HP-ZBook-15:~/gr-iridium$ sudo iridium-extractor -D 4 examples/hackrf.conf > output.bits
Traceback (most recent call last):
  File "/usr/local/bin/iridium-extractor", line 357, in <module>
    tb = iridium.iridium_extractor_flowgraph.FlowGraph(center_frequency=center, sample_rate=sample_rate, decimation=decimation, 
  File "/usr/local/lib/python3.10/dist-packages/iridium/iridium_extractor_flowgraph.py", line 150, in __init__
    source = osmosdr.source(args=d['device_args'])
RuntimeError: logger_set_level: Bad level type.
corrupted size vs. prev_size in fastbins
Aborted

[image]

schneider42, 2024-07-27:
Is it possible that you have multiple versions of gr-iridium installed? can you delete the whole `/usr/local/lib/python3.10/dist-packages/iridium` folder and try again?

ashwini1890, 2024-07-30:
no i have only 1 version of gr-iridium. if i will delete this the iridium module will be deleted 
[image]