muccc/gr-iridium #209: <string>:3: SyntaxWarning: invalid escape sequence '\W' error during CMAKE process

RobertSwirsky, 2025-01-17:
```
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.6")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.12.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/include (found version "2.11.1")
-- Using install prefix: /usr/local
-- Building for version: v1.0.0-51-geed634ca / 1.0.0git
-- No C++ unit tests... skipping
<string>:3: SyntaxWarning: invalid escape sequence '\W'
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- PYTHON and GRC components are enabled
-- CMAKE_INSTALL_PREFIX: /usr/local
-- GR_PYTHON_DIR: lib/python3.12/dist-packages
-- GR_PYTHON_DIR: lib/python3.12/dist-packages
-- Python checking for pygccxml - found
<string>:3: SyntaxWarning: invalid escape sequence '\W'
```

when following the CMAKE build instructions. Everything seems to work, and this may be a bug elsewhere, but I thought I'd note it here in case others are seeing it and are wondering what it is.

Here's the branch and commit I'm building from 
```
$ git log
commit eed634ca388394fcfd2531187aa77537372519db (HEAD -> master, origin/master, origin/HEAD)
Author: Sec <sec@42.org>
Date:   Wed Nov 20 20:37:20 2024 +0100

    iridium-extractor: add generate-sigmf-meta feature (#207)

```

The command I'm using is 

`$ cmake -B build`

after I've cloned the repo

schneider42, 2025-01-19:
Thanks for reporting. That is a bug in GNU Radio and has been fixed already: https://github.com/gnuradio/gnuradio/pull/7037

Might take some time until it is distributed by your OS distribution.

closed 2025-01-19 by schneider42

RobertSwirsky, 2025-01-20:
Thanks! I see it's harmless, so I won't worry about it. (I build GnuRadio from source here.)