muccc/gr-iridium #211: TDOA with Iridium Signals? A first rough test shows unexpected results using iridium-extractor.

RobertSwirsky, 2025-01-26:
Reading the code, it seems like if you're using a USRP with a PPS clock, it will use the rx_time + samples to make the time based on a calculation from sample count.

So I set up a quick experiment: I took two genuine Ettus B-200s, both driven off the same GPSDO 10Mhz and PPS., with the timing signals split from a proper 10 MHz distribution amp, etc.  There's a Tallisman Iridium antenna on the roof of our building and we split the signals with a proper splitter to both of the USRP B200s.

[image]

I have two computers with the exact same versions of everything running iridium-extractor.  I wrote a script to compare the "time" field in the third field from each of the captures in the RAW: file, only looking at "100%" decodes. I expected to see identical times on each of them.  (I'm also ignoring "179" length decodes, and making sure all the bits match, too, before comparing the time.)

Here's the end of a recent run. As you can see it's not a conisitant lag, and sometimes packets are ahead and behind, ranging from 100 nanoseconds to 1000 nanoseconds.  The average time difference at the end of this run was 296 nanoseconds (0.000296 seconds).

Any ideas as to what's happening? I was hoping these would line up, and then I'd move on to doing sub-sample timing but I need to get the gross time measurements to be much closer/consistent before I can hope to get reliable results with separate receivers in separate locations.

Thanks!

[...]
```
**** Matched Time difference == -0.001000, 3401 records left (record len 432) [Comparing  0633778.4164 and 0634778.4179]
**** Matched Time difference == -0.000100, 3358 records left (record len 432) [Comparing  0634858.4166 and 0635858.4172]
**** Matched Time difference == 0.000700, 3295 records left (record len 432) [Comparing  0637288.4188 and 0638288.4186]
**** Matched Time difference == 0.000500, 3247 records left (record len 432) [Comparing  0638908.4206 and 0639908.4206]
**** Matched Time difference == 0.000800, 3203 records left (record len 432) [Comparing  0642148.4255 and 0643148.4252]
**** Matched Time difference == 0.000000, 3180 records left (record len 432) [Comparing  0643768.4270 and 0644768.4275]
**** Matched Time difference == 0.000000, 3142 records left (record len 432) [Comparing  0645388.4296 and 0646388.4301]
**** Matched Time difference == -0.000300, 3115 records left (record len 432) [Comparing  0645928.4298 and 0646928.4306]
**** Matched Time difference == 0.000000, 3028 records left (record len 432) [Comparing  0648088.4344 and 0649088.4349]
**** Matched Time difference == -0.000800, 3002 records left (record len 432) [Comparing  0648898.4358 and 0649898.4371]
**** Matched Time difference == 0.000700, 2968 records left (record len 432) [Comparing  0649708.4385 and 0650708.4383]
**** Matched Time difference == 0.001000, 2919 records left (record len 432) [Comparing  0651058.4417 and 0652058.4412]
**** Matched Time difference == -0.000000, 2861 records left (record len 432) [Comparing  0653218.4463 and 0654218.4468]
**** Matched Time difference == 0.000500, 2844 records left (record len 432) [Comparing  0654028.4495 and 0655028.4495]
**** Matched Time difference == 0.000700, 2814 records left (record len 432) [Comparing  0655378.4532 and 0656378.4530]
**** Matched Time difference == 0.000500, 2785 records left (record len 432) [Comparing  0656188.4549 and 0657188.4549]
**** Matched Time difference == 0.000300, 2682 records left (record len 432) [Comparing  0658348.4616 and 0659348.4618]
**** Matched Time difference == -0.000300, 2665 records left (record len 432) [Comparing  0658888.4628 and 0659888.4636]
**** Matched Time difference == 0.000700, 2632 records left (record len 432) [Comparing  0659698.4655 and 0660698.4653]
**** Matched Time difference == -0.000300, 2519 records left (record len 432) [Comparing  0660778.4686 and 0661778.4694]
**** Matched Time difference == -0.000200, 2446 records left (record len 432) [Comparing  0661588.4726 and 0662588.4733]
**** Matched Time difference == 0.000200, 2414 records left (record len 432) [Comparing  0661858.4732 and 0662858.4735]
**** Matched Time difference == 0.000500, 2352 records left (record len 432) [Comparing  0662668.4764 and 0663668.4764]
**** Matched Time difference == 0.000200, 2233 records left (record len 432) [Comparing  0663748.4815 and 0664748.4818]
**** Matched Time difference == -0.000000, 2114 records left (record len 432) [Comparing  0664828.4844 and 0665828.4849]
**** Matched Time difference == 0.000000, 2078 records left (record len 432) [Comparing  0665368.4856 and 0666368.4861]
**** Matched Time difference == 0.001000, 2039 records left (record len 432) [Comparing  0666178.4896 and 0667178.4891]
**** Matched Time difference == 0.000800, 2001 records left (record len 432) [Comparing  0666988.4931 and 0667988.4928]
**** Matched Time difference == 0.000300, 1965 records left (record len 432) [Comparing  0667528.4959 and 0668528.4961]
**** Matched Time difference == 0.000300, 1916 records left (record len 432) [Comparing  0668338.4986 and 0669338.4988]
**** Matched Time difference == -0.000300, 1858 records left (record len 432) [Comparing  0669148.5010 and 0670148.5018]
**** Matched Time difference == -0.000000, 1739 records left (record len 432) [Comparing  0670498.5075 and 0671498.5080]
**** Matched Time difference == 0.000200, 1629 records left (record len 432) [Comparing  0671848.5140 and 0672848.5143]
**** Matched Time difference == 0.000500, 1628 records left (record len 432) [Comparing  0671848.5146 and 0672848.5146]
**** Matched Time difference == 0.000200, 1580 records left (record len 432) [Comparing  0672658.5175 and 0673658.5178]
**** Matched Time difference == 0.000300, 1527 records left (record len 432) [Comparing  0673468.5215 and 0674468.5217]
**** Matched Time difference == 0.000500, 1498 records left (record len 432) [Comparing  0674008.5248 and 0675008.5248]
**** Matched Time difference == 0.000500, 1472 records left (record len 432) [Comparing  0674818.5280 and 0675818.5280]
**** Matched Time difference == 0.000000, 1447 records left (record len 432) [Comparing  0675628.5320 and 0676628.5325]
**** Matched Time difference == 0.000500, 1431 records left (record len 432) [Comparing  0676168.5345 and 0677168.5345]
**** Matched Time difference == 0.000300, 1414 records left (record len 432) [Comparing  0676978.5388 and 0677978.5390]
**** Matched Time difference == 0.000800, 1304 records left (record len 432) [Comparing  0678328.5463 and 0679328.5460]
**** Matched Time difference == 0.000200, 1172 records left (record len 432) [Comparing  0681298.5620 and 0682298.5623]
**** Matched Time difference == 0.000000, 1111 records left (record len 432) [Comparing  0682648.5681 and 0683648.5686]
**** Matched Time difference == 0.001300, 1090 records left (record len 432) [Comparing  0683098.5728 and 0684098.5720]
**** Matched Time difference == 0.000300, 1074 records left (record len 432) [Comparing  0683458.5741 and 0684458.5743]
**** Matched Time difference == -0.000100, 1014 records left (record len 432) [Comparing  0685348.5862 and 0686348.5868]
**** Matched Time difference == -0.000300, 941 records left (record len 432) [Comparing  0686968.5947 and 0687968.5955]
**** Matched Time difference == 0.000300, 901 records left (record len 432) [Comparing  0687778.5997 and 0688778.5999]
**** Matched Time difference == 0.000000, 865 records left (record len 432) [Comparing  0689128.6080 and 0690128.6085]
**** Matched Time difference == 0.000000, 864 records left (record len 432) [Comparing  0689128.6090 and 0690128.6095]
**** Matched Time difference == 0.000500, 857 records left (record len 432) [Comparing  0689398.6102 and 0690398.6102]
**** Matched Time difference == 0.000300, 851 records left (record len 432) [Comparing  0689938.6138 and 0690938.6140]
**** Matched Time difference == 0.000000, 808 records left (record len 432) [Comparing  0691288.6226 and 0692288.6231]
**** Matched Time difference == 0.000700, 803 records left (record len 432) [Comparing  0691558.6245 and 0692558.6243]
**** Matched Time difference == -0.000300, 791 records left (record len 432) [Comparing  0692098.6273 and 0693098.6281]
**** Matched Time difference == 0.000200, 763 records left (record len 432) [Comparing  0693718.6388 and 0694718.6391]
**** Matched Time difference == 0.000500, 750 records left (record len 432) [Comparing  0694258.6422 and 0695258.6422]
[... cut ...]
**** Matched Time difference == 0.000300, 242 records left (record len 432) [Comparing  0704788.7194 and 0705788.7196]
**** Matched Time difference == 0.001300, 150 records left (record len 432) [Comparing  0706048.7304 and 0707048.7296]
**** Matched Time difference == 0.000000, 115 records left (record len 432) [Comparing  0706408.7319 and 0707408.7324]
**** Matched Time difference == 0.000500, 43 records left (record len 432) [Comparing  0707218.7384 and 0708218.7384]
Average time difference = 0.000296

```

Sec42, 2025-02-03:
That sounds unexpected to me. We will have to do some more investigation to give a better answer.

RobertSwirsky, 2025-02-03:
I noticed during the Sec and Schneider talk at CCC (https://media.ccc.de/v/38c3-investigating-the-iridium-satellite-network) they mentioned that the USRP seems to not timestamp packets accurately and there's a delay. I've never noticed this in my work with USRP B200s and 300 series devices. But if there's a delay and it's not consistent, this could be an issue. I'm going to do some more measurements today to make sure that the timing between the devices is identical (again with a common external GPS-locked 10Mhz and PPS feeding both of my USRPs.)

Sec42, 2025-02-04:
The delay we mentioned in the talk should be deterministic (dependent on sample rate). The delay should be similar to what is known for the N210: http://ianbuckley.net/N210-rx-dsp-delay.txt

I am not an expert on this, but I suspect, running a test with a generated reference signal into a splitter with two usrp's could test this issue.

schneider42, 2025-02-04:
Honestly an average of ~300 ns is not bad. We only determine the timestamp based on 12 symbols from the unique word and each symbol of these is 40 us long. I believe we'd need to correlate against more symbols to get more accurate. That certainly is doable with reasonable effort. So far we simply were not in a position where this made sense. TDOAing transmitters obviously would be an application of course.

RobertSwirsky, 2025-02-04:
Thank you very much for your answers. 

We ran a test with my own 1GHz signal generated in the lab, and time measurements on a pair of USRPs from the same antenna (split) on the same 10MHz clock and PPS sent through distribution amps. Timing was within 1 sample.

Based on your comments I will try to simply dump a longer sample at key burst-detection times and cross-corellate between the two devices.

I have another question: 

I presume, with my Tallisman Iridium antenna sitting on our roof....

[image]<br><caption>Here it is before it went on the roof</caption>

....that we're receiving multiple satellites simutaneously in the pager part of the band (where the new proprietary timing data is, too?) but in the "voice and data" area, we are only going to be receiving from whatever satellite's spotbeam is beaming down to us at any one time. Is that correct? Also, do you know what the extent of the spot beams are? (How much area do they cover).

If we were to do practical TDOA using these signals as "Signals of Opportunity" (and not decoding any of the proprietary data), I'd presume we'd have to look only at the "pager" part so we can get multiple receivers in multiple locations looking at the same satellites  because we're not going to be able to see multiple satellites in voice/data band at same time, and the range of the voice/data band is limited to the size of the "spot"

RobertSwirsky, 2025-02-11:
Update! I switched to a better distribution amp for the PPS (I was using a known good one for the 10Mhz) and the average difference between the two of them dropped to an average of 0.000043 millisconds (43 nanoseconds). 

Here's the end of a recent run comparing packets that were identical between the two USRPs (again, I started them a second apart, so we corrected for the 1000 millisecond difference between them)

`		**** Matched Time difference == -0.000300, 1268 records left (record len 432) [Comparing  0611289.4845 and 0610289.4848]
		**** Matched Time difference == -0.000500, 1226 records left (record len 432) [Comparing  0612099.4701 and 0611099.4706]
		**** Matched Time difference == 0.000000, 1179 records left (record len 432) [Comparing  0613539.4435 and 0612539.4435]
		**** Matched Time difference == -0.000300, 1129 records left (record len 432) [Comparing  0615609.4054 and 0614609.4057]
		**** Matched Time difference == 0.000200, 1115 records left (record len 432) [Comparing  0616419.3902 and 0615419.3900]
		**** Matched Time difference == 0.000000, 1082 records left (record len 432) [Comparing  0617859.3644 and 0616859.3644]
		**** Matched Time difference == 0.000500, 1068 records left (record len 432) [Comparing  0619929.3263 and 0618929.3258]
		**** Matched Time difference == 0.000200, 968 records left (record len 432) [Comparing  0624249.2485 and 0623249.2483]
		**** Matched Time difference == 0.000200, 964 records left (record len 432) [Comparing  0624339.2481 and 0623339.2479]
		**** Matched Time difference == 0.000000, 755 records left (record len 432) [Comparing  0630819.1306 and 0629819.1306]
		**** Matched Time difference == 0.000500, 726 records left (record len 432) [Comparing  0632889.0944 and 0631889.0939]
		**** Matched Time difference == -0.000500, 702 records left (record len 432) [Comparing  0635139.0543 and 0634139.0548]
		**** Matched Time difference == 0.000300, 614 records left (record len 432) [Comparing  0637209.0179 and 0636209.0176]
		**** Matched Time difference == 0.000000, 537 records left (record len 432) [Comparing  0639458.9791 and 0638458.9791]
		**** Matched Time difference == -0.000300, 444 records left (record len 432) [Comparing  0641528.9431 and 0640528.9434]
		**** Matched Time difference == 0.000300, 373 records left (record len 432) [Comparing  0643778.9046 and 0642778.9043]
		**** Matched Time difference == -0.000200, 324 records left (record len 432) [Comparing  0644858.8867 and 0643858.8869]
		**** Matched Time difference == 0.000800, 310 records left (record len 432) [Comparing  0645848.8694 and 0644848.8686]
		**** Matched Time difference == -0.000200, 309 records left (record len 432) [Comparing  0645938.8685 and 0644938.8687]
		**** Matched Time difference == -0.000200, 280 records left (record len 432) [Comparing  0647018.8493 and 0646018.8495]
		**** Matched Time difference == -0.000500, 258 records left (record len 432) [Comparing  0648098.8306 and 0647098.8311]
		**** Matched Time difference == 0.000200, 185 records left (record len 432) [Comparing  0650168.7956 and 0649168.7954]
		**** Matched Time difference == 0.000000, 150 records left (record len 432) [Comparing  0651338.7768 and 0650338.7768]
		**** Matched Time difference == 0.000200, 111 records left (record len 432) [Comparing  0652418.7576 and 0651418.7574]
`

Sec42, 2025-02-16:
To answer your question: All iridium signals are sent via the same spot beams, so you will see all types of packets from multiple satellites at the same time. The IMS/MSG (pager) and ITL (time&location) are sent with higher transmit power so you're more likely to see them.
For TDOA correlation the IMS/MSG signals are probably best due to another reason: they start with the iridium access control word (12 symbols) directly followed by the messaging header (16 symbols) which will give you a longer known signal to correlate against.