muccc/gr-iridium #212: sampling rate cnfiguration issue with soapy source

yuval174, 2025-01-29:
hey, I'm used this project a few month ago and it worked great so thank you first!
i tried using it again on a new of install ubuntu 22.04 (same laptop and same RTL-SDR) and i run into an issue.
i use gnu-radio 3.10.7 and gr-osmosdr isn't part of the core gnuradio any more, further more when i tried installing it i run to some problem with loading gnuradio companion flowgraph. this means i would rather use soapy source since it is part of the core blocks if it's possible.

the issue is  that when using soapy source for RTL-SDR it limit the sampling rate to a very specific list i don't have it right now but i can find it tomorrow if it's essential, it contain sampling rate like: 1.024Msps, 2.048 Msps, 3.2Msps and some other value. the point it that i cant select simply 2Msps or other round value. this leads to the fact that the sampling rate is not a multiple of 100Ksps like it is required by iridium-extractor.

is there a workaround this so it will still be possible to use source blocks of soapysdr, and avoid installing gr-osmosdr?

thank you very much

Sec42, 2025-02-03:
To my knowledge that is an issue with soapy -- the application can only use the sampling rates that soapy allows. It looks like currently you need to switch back to osmosdr for rtl-sdr to make it work.

I'll leave this open for schneider to investigate and/or correct me :-)

yuval174, 2025-02-05:
thank you for the response, I guess I will try to resolve the issue with gr-osmosdr breaking gnuradio. maybe going back to earlier version of gnuradio will fix this

schneider42, 2025-02-16:
To my knowledge this is an issue with https://packages.ubuntu.com/jammy/soapysdr-module-rtlsdr shipped with 22.04. You'd need 0.3.3: https://github.com/muccc/gr-iridium/issues/159#issuecomment-1232946781

24.04 seems to be fine: https://packages.ubuntu.com/noble/soapysdr-module-rtlsdr

yuval174, 2025-02-17:
hey thanks for providing more solutions, I currently worked around this by doing some order in my machine and going back to gr-osmosdr as @Sec42  suggested.
I completely uninstalled gnuradio, librtlsdr, gr-osmosdr and all of their duplication i had.
installed gnuradio with "sudo apt install gnuradio" to get 3.10.1 and then building manually from source gr-osmosdr.
this worked, so this Issue is resolved as far as I'm concern :).
thank you again

schneider42, 2025-02-17:
Great that you got it working. Closing this issue then :)

closed 2025-02-17 by schneider42