muccc/gr-iridium #213: Issue with Channel Number Modification in iridium-qpsk-demod Module

xxs0215, 2025-02-12:
Hello Dear Developers,

When I modify the number of channels in the iridium-qpsk-demod module, the cpdu input ports do not increase to match the new channel count. This causes an error when running the flow graph. 

[image]

[image]

Could you please take a look at this? Thank you very much for your time and attention. I look forward to your guidance on how to resolve this issue. 

Best Regards.

schneider42, 2025-02-16:
Looks like a bug in GRC which doesn't correctly creates new ports in the UI. Might be related to it being a port of domain `message`.  Please consider filing a bug upstream.

schneider42, 2025-03-29:
Apparently a typo in our code :)

closed 2025-03-29 by schneider42