muccc/gr-iridium #215: No timestamp found in filename

fannoname, 2025-02-24:
I'm new to gr-iridium and the Iridium Toolkit.When I'm working with raw data，I ran the following command to decode the raw data:
 **python3 iridium-extractor -c 1626000000 -r 10000000 --file-info 1732051200 -f cu8 /mnt/hgfs/caiji/1200.dat >output.bits**
and ran the following command next:
**python3 iridium-parser.py -p ~/gr-iridium/apps/output.bits**
Then the following warning appeared:
**Warning: no timestamp found in filename [0.0 l/s] filtered:100%**
And the resulting output.parsed file is in the form below:
**IBC: u-1732051200 000000438.7346 1623829102  85% -18.00|-067.27|18.03 131 DL bc:0 sat:028 cell:08 0 slot:0 sv_blkn:0 aq_cl:1111111111111111 aq_sb:24 aq_ch:2 00 0000 time:2024-11-20T02:20:36.89Z                      [7 100111000101000010100000001010010000010]**
The prefix is not p- but u-.
I would like to know if this result will have an effect on the following commands, and if so, how to fix it.
**python3 reassembler.py -i output.parsed -m satmap**
Where my timestamp should have been 1732051200.

Sec42, 2025-02-24:
If your intention is to provide a timestamp for the start of the recording, you will have to use a format like `--file-info i-1732051200-t1`.

fannoname, 2025-02-25:
This method solved my problem, thank you very much for your help！